Gaboze-Pocaio/Gaboze-Pocaio-Round-2---Perfect #4: PCB Specs / DIY Instructions?

biergaizi, 2019-01-21:
Hello.

First, thanks for designing this awesome board for the community. I was impressed by it and was planning to build it at home by the end 2018. But today, I've just discovered the design of Round 2 is now available, so it seems the best option for me is to start directly with Round 2? However, the project wiki of GabozePocaio-Round1 has detailed information for building the kit from scratch at home, but currently there's no information for Round2.

Could you give me some quick tips on building it? Specifically...

* What is the manufacturing specs of the new PCB? Is it the same one found the Round 1?
 
* If I reuse the build instructions in Round 1, what are the differences that should be taken into account? I know I should order a new PCB and order the parts by referencing to the new BOM. But how about other things that should be noted when building it?

* How should be enclosure be modded?

Thanks.

32teeth, 2019-01-21:
> * What are the manufacturing specs of the new PCB? Is it the same one found the Round 1?
Yes, same, 1mm thickness

> * If I reuse the build instructions in Round 1, what are the differences that should be taken into account? I know I should order a new PCB and order the parts by referencing to the new BOM. But how about other things that should be noted when building it?
https://github.com/32teeth/GabozePocaio-Round2/wiki/Pi-Mounting

> * How should be enclosure be modded?
https://github.com/32teeth/GabozePocaio-Round2/wiki/Preperation

biergaizi, 2019-01-21:
> Yes, same, 1mm thickness

What about the size? Is it still 115.5 x 71.9?

biergaizi, 2019-01-21:
I've submitted my order to the PCB shop. Thanks again for the quick response! :tada:

closed 2019-01-21 by biergaizi