Gaboze-Pocaio/Gaboze-Pocaio-Round-2---Perfect #5: Spot D1

Vygr10565, 2019-05-10:
Hey 32teeth,

First of all thanks for this awesome project! I was wondering which transistor should be on the D1 spot, circled in blue underneath the Raspberry Pi.]

[image]

32teeth, 2019-05-10:
Designator D1 is a Diode

|  Designator    |   Description   |   MPN   |   Purchase Link   |
| ---- | ---- | ---- | ---- |
|  D1    |   DIODE SCHOTTKY 20V 1A SOD123FL   |   SM5817PL-TP   |   [DigiKey](https://www.digikey.ca/product-detail/en/micro-commercial-co/SM5817PL-TP/SM5817PL-TPMSCT-ND/1793410)   |

[More Here](https://github.com/32teeth/GabozePocaio-Round2/blob/master/bom/GabozePocaioRound2.csv)

closed 2019-05-10 by 32teeth

Vygr10565, 2019-05-10:
Will the Raspberry Pi startup without the diode?