Gaboze-Pocaio/Gaboze-Pocaio-Round-2---Perfect #6: What is the Correct Way to Install the 40-Pin Header and the Screen?

biergaizi, 2019-06-10:
Hello.

I finally started to assemble the board. I was able to install all the components and the system booted successfully. But I found I've solder the pin header directly, and I found the pins would penetrate to the other side of the board, together with the solder joints, the screen is impossible to fit on the board, so it's not possible to install the board inside the enclosure. I've read through all the resources about GabozePocaio, but it seems none of them mentioned how to solder the 40-Pin header for the Raspberry Pi mount.

I tried to use a wire-cutting but I found the pins are too short to cut, I also tried to grind the pins out using a Dremel, but I found it's very easy to damage the solder mask in this process. Currently my result of the experiment is three dead boards.

My questions are:

1. What is the correct way to install the header and the screen? Is the pin header in the project BOM a non-standard low-profile version, or is there a special method used here to install it? I've reviewed the spec of TLW-120-06-T-D and I didn't find anything special, so I just used a generic pin header in my toolbox, is it possible that I've missed something?

2. How is the screen installed on the board? How is the proper clearance possible to achieve despite the solder joints between the board and the screen? None of the picture shows what's behind the screen. 

Thanks again for creating this cool project.

32teeth, 2019-06-10:
[image]
[image]
[image]
[image]
[image]
[image]
[image]

biergaizi, 2019-06-11:
Awesome, thanks!

closed 2019-06-11 by biergaizi