Gaboze-Pocaio/Gaboze-Pocaio-Round-2---Perfect #7: Correct/Best way to remove Raspberry Pi from Gaboze Pocaio?

GameDragon2k, 2019-06-24:
Hey 32teeth.

I've seem to run in an issue with my device. Everything was assembled and running perfectly, but it seems somewhere in the process of closing everything into the case, something got shorted out. Still can't confirm, but I believe that the Pi I am using is dead now.

What steps should I take to remove this Pi from the Gaboze Pocaio? Everything is already soldered down of course.

32teeth, 2019-06-28:
https://github.com/32teeth/GabozePocaio-Round2/releases

closed 2019-06-28 by 32teeth

GameDragon2k, 2019-07-01:
I'm confused. This doesn't really answer the question. My image was already installed and working prior, games and everything. Screen no longer turns on (no white screen) but the TP4056 seems to be getting power correctly. Tried reflowing both and no avail.

The Pi on the other hand just instantly overheats once powered, so I'm guessing that something is wrong with it. I have other Pis to replace it with, but I have no idea how to go about replacing it on the Gaboze Pocaio itself.

32teeth, 2019-07-03:
Please post a picture of the Pi Soldering

biergaizi, 2020-04-02:
@GameDragon2k My experience is that it's nearly impossible to remove the Raspberry Pi from the board by using manual tools to desolder each pin individually, there's a high chance that you are going to damage the board in this process, yet the solder still stuck in its place. Please refer to the book *Quality Hand Soldering and Circuit Board Repair* by H. (Ted) Smith, 6th edition, page 90, which explained the problem of the difficulty of desoldering components on a double-side PCB.

The proper way of doing it is purchasing a vacuum desoldering gun, which is able to reflow the solder and removing it simultaneously. You simply add fresh solder to all joints with a soldering iron, then you add fresh flux to all joints. Finally, you use a vacuum desoldering gun on the joint and press the trigger to remove all solder. If some pins are stuck at one side of the pad, repeat the process - add fresh solder and remove it with a desolder gun, after the solder joint is heated with the gun, bend the pin to the opposite direction to free it. 

if you don't have a vacuum desoldering gun, either buying/borrowing one for $100, or accepting a greater risk of damaging your board. To minimized the chance of board damage, don't even think about removing it with a soldering iron if you don't know what you are doing. The method of the lowest risk of damage is *destructive removal*. Kill the pin header, kill the dead Raspberry Pi, but your board is saved. You can use a wire cutter to reach the plastic header under the Raspberry Pi, and cut the connectors into pieces, so the Raspberry Pi is freed from the board. If the headers under the Pi is difficult to reach, I imagine that using a Dremel with a cutting wheel (angle grinder) should also work (be careful not to damage the board underneath). Once the Pi is destructively removed, you use a wire cutter to destroy all plastic tabs on the header, adding some fresh solder to pad, heating it with a soldering iron and pulling it outwards on the other side simultaneously (a board holder is needed) to remove one pin, repeat it for 40 times to remove everything. Never force anything out, or you will damage the pad. If you feel that it refuses to go out, adding more fresh solder and flux to allow the joint to reflow completely. 

You see, reflowing electronics is a painstaking process and it requires proper tools. Sometimes, a PhD in electronics engineering does nothing to help you removing a connector from a motherboard. If you are not sure how to proceed, I recommend you to find an electronic repair shop to pay someone to do it, a shop that does TV/home appliance/laptop/smartphone repair should be able to do the job.