Gaboze-Pocaio/Gaboze-Pocaio-Round-2---Perfect #10: White lines across screen

SpyderGiant, 2019-12-31:
I just finished building my gaboze, but I’m having an issue with the screen having white lines across just the top half. Any idea what the problem might be or if I’ve just got a bad screen?
[image]
[image]
[image]

32teeth, 2019-12-31:
Hmm, I would definitely check the soldering of the LCD ribbon

This to me looks like “noise”

SpyderGiant, 2019-12-31:
Just realized I didn’t post the right pic of the ribbon I soldered. 
[image]

SpyderGiant, 2020-01-19:
Solved. Got a new screen from Ali seller and rebuilt for a 3rd time. Working now
[image]

closed 2020-01-19 by SpyderGiant