Gaboze-Pocaio/Gaboze-Pocaio-Round-2---Perfect #12: What is the Consideration Behind the Unusually Low AC-coupling Capacitor C2 = 220 pF?

biergaizi, 2020-04-06:
What is the design consideration behind LM4875's input AC coupling capacitor C2? According to the datasheet, the AC coupling capacitor C2 and the input impedance R of LM4875 forms a highpass RC filter. At minimum volume, R = 110kΩ, which means the 3 dB point of the filter is 1 / (2 x π x 110 kΩ x 220 pF) = 6.576 kHz, in other words, the low frequency components, especially bass, is completely blocked by the capacitor. As the volume increases, the 3 dB point moves higher and higher, and cut off more and more frequencies at the lower side. 

Personally, I found the audio quality of the headphone output jack is greatly improved after replacing C2 with a 0.1 uF capacitor.

What is the specific reason to use a 220 pF coupling capacitor here? Is it meant to avoid damaging the GPIO port of the Raspberry Pi?

32teeth, 2020-04-06:
> Recommended From TI Datasheet
http://www.ti.com/lit/ds/symlink/lm4875.pdf

closed 2020-04-06 by 32teeth

biergaizi, 2020-04-06:
It's not what TI recommended. TI recommends a 0.22 uF capacitor, which is 220 **nF** (or 220e-9), not 220 pF (220e-12). I think that the issue should be reopened.

biergaizi, 2020-04-06:
I opened Pull Request #13 with a fix to this problem.