Gaboze-Pocaio/Gaboze-Pocaio-Round-2---Perfect #15: 51p screen not available

mlopezmad, 2020-09-10:
Hi friend!
I hope you are good
I know that it has been commented, but I have two complete kits to assemble and I am missing the screen, there are only 37 pin screens. 

 I can't find a provider with 51p screens. 

Could you help us to adapt the PCB footprint to the new screen 37 pin one?  

Thank you very much!!!

whqsz, 2020-12-16:
The supplier has stopped shipping the 51PIN version of the 2.6-inch SCREEN that supports SPI. It is not easy to find

32teeth, 2020-12-19:
@mlopezmad and @whqsz thanks for the comment
we are currently working on modifications for our current project 

it's a 45pin 0.5mm FPC Connector for an IPS screen
We have only adapted it for the Espressif ESP32 chipset, not the Raspberry Pi

[image]

[Sponsorship Retro ESP32](https://github.com/sponsors/retro-esp32)

Eleecho, 2021-04-29:
Darn, I read this too late. 

I am sitting with 3 kits worth of parts and boards and three 37 Pin screens as that's where the link leads to in your BOM.

My fault for not digging deeper and checking parts, but could you update the BOM cell to remove the link and say "51Pin 2.6" TFT unavailable" or something in case others attempt to source parts and build?

Cheers,

Aaron